magicbug/Cloudlog #1011: Grid square map not update when import ADIF

EA8CXN, 2021-04-30:
Hi!
I recently made a fresh install of my cloudlog into a different server and I check that my previous QSO imported by ADIF don't appear on the MAP.
They are on the logbook, and on my VUCC appears also with the Grid Locator ok, but the grid square map is not marked as worker or confirmed.

I miss to do something after import the ADIF?

Thanks, EA8CXN, César

EA8CXN, 2021-04-30:
After continue checking I see what is the problem.
When I export ADIF from my old Cloudlog, the SAT Name and SAT Mode didn't export, so after update my entry into the new cloudlog, the gridsquare appear confirmed.

I don't know if it is possible to made a full data export.

Thanks

magicbug, 2021-05-02:
I'm not sure what version, but for sat fields to missing somethings one a bit weird as I don't remember any release where they weren't available? depending on how old the install is you can take the whole logbook table out of MySQL. 

Be Interested to know some more as quite puzzled.

magicbug, 2021-05-02:
Be good to see the ADIF export and any info you have on what code version you might be using as I'm not aware of any issues with those parts.

closed 2021-05-09 by magicbug