magicbug/Cloudlog #1014: Database Error

DK2KL, 2021-05-01:
Hello,

finally I've got cloudlog installed on my QNAP TS-220 with no errors but when I call the main page, it'll run in the error:

A PHP Error was encountered

Severity: Warning

Message: mysqli::set_charset(): Error executing query

Filename: mysqli/mysqli_driver.php

Line Number: 275
A PHP Error was encountered

Severity: Warning

Message: Cannot modify header information - headers already sent by (output started at /share/MD0_DATA/.qpkg/cloudlog/system/core/Exceptions.php:271)

Filename: core/Common.php

Line Number: 570
A Database Error Occurred

Unable to set client connection character set: utf8mb4

Filename: core/CodeIgniter.php

Line Number: 518

I've changed in the database the collation to UTF8_General_mysql500_ci 'cause utf8mb4 is not available but it's running again into this issue.

I'll be very glad, if this problem can be solved.

Regards
Jens

magicbug, 2021-05-02:
Jens,

What verison of PHP & MySQL?

Peter.

DK2KL, 2021-05-02:
> What version of PHP & MySQL?

Hello Peter,

thank You for responding.

My NAS is running with PHP 5.6.40, MySQL 5.1.73 server and mysqlnd 
5.0.11 database-client.

Vy 73

Jens

Jens Knorr - DK2KL

magicbug, 2021-05-02:
Sadly as per the readme it doesn't meet the minimum spec for Cloudlog to work

closed 2021-05-02 by magicbug