magicbug/Cloudlog #1016: Option to show QSOs from all logbooks

thelovebug, 2021-05-03:
Currently, the QSO list only shows QSOs from the active logbook.  Tt would be good to have an option somewhere (perhaps as a tick-box in the Station Locations page) to allow the QSO list to show all entries from all logbooks.

Use cases:

- The home page of a Cloudlog instance can be shared with others to be able to see a list of recent QSOs.  If you're working multiple stations (for example yesterday when I was working contacts from two different locations whilst portable), setting one specific logbook limits the list to exclude those from others.
- When logged in, I want to have the option of seeing _all_ QSOs from a given time frame, regardless of the logbook they were logged in.

magicbug, 2021-05-03:
Its coming, its taking a bit longer than expected but will come with the "Logbooks" feature, once that's implemented can do things like the above, but better to hold as it will add more work if not done together.

thelovebug, 2021-05-07:
Actually, I've just realised that the Cloudlog home page, and the Logbooks page, are different.  Logbooks actually gives me what I was asking for, and I notice that this is available to visiting (not-authenticated) users also.  On that basis, I'm happy to close this off as my "issue" is, in fact, a non-issue after all. 😄

closed 2021-05-07 by thelovebug