magicbug/Cloudlog #1023: Database table is not properly deployed

BH6AOV, 2021-05-07:
**Describe the bug**
After I installed by http://domain/install, database is wrong on the web page：
```text
A Database Error Occurred
Error Number: 1146
Table 'cloudlog_db.options' doesn't exist
SELECT * FROM `options` WHERE `autoload` = 'yes'
Filename: models/Options_model.php
Line Number: 19
```

**To Reproduce**
1.Install LNMP
2.Set database with wiki's guide
3.Install Cloudlog from http://domain/install
4.Open http://domain
5.Saw the error mentioned above.

**Expected behaviour**
normal operation

**Screenshots**
[image]

**Server**
 - OS: Ubuntu 18.04 LTS Server
 - WebServer: nginx/1.14.0 (Ubuntu)
 - Database: mysql  Ver 14.14 Distrib 5.7.33, for Linux (x86_64) using  EditLine wrapper
 - php: PHP 8.0.5 (cli) (built: May  3 2021 11:30:38) ( NTS )

**Additional context**
I tried to manually add an error, but there have been other problems, I hope I can check the problem.I need CloudLog very much.

BH6AOV, 2021-05-07:
This is a database list after installation.[image]

magicbug, 2021-05-08:
Is this a fresh install? its possibly relating to PHP8 (untested) looks like migrations aren't running.. 

My only suggestion at the moment is use PHP 7.x branch although I know its unpopular choice, will try spend some time getting Cloudlog PHP8 ready soon.

BH6AOV, 2021-05-08:
> Is this a fresh install? its possibly relating to PHP8 (untested) looks like migrations aren't running..
> 
> My only suggestion at the moment is use PHP 7.x branch although I know its unpopular choice, will try spend some time getting Cloudlog PHP8 ready soon.

Yes, this is a fresh install.
This problem are caused by PHP8, maybe.
I will try remove PHP8 and install PHP 7.3.I will purge all files and fresh reinstall again.If still have this problem, I will feedback directly here.

BH6AOV, 2021-05-08:
@magicbug OK, now it is running, but I can't login while I login the demo account then will give me 404 Not Found.Do you know what happend?
The 404 Page's address is https://domain/index.php/user/login
[image]

magicbug, 2021-05-08:
Looks like a server config issue, I'd check your logs to see what's up.

BH6AOV, 2021-05-09:
> Looks like a server config issue, I'd check your logs to see what's up.

uhhh...Which log files do you want?Nginx log Or install log?

magicbug, 2021-05-09:
As this is a issue with server configs rather than Cloudlog I'll change it over to a discussion so you can get help with setting up Nginx.

closed 2021-05-09 by magicbug