magicbug/Cloudlog #1037: Database error when visiting US Counties awards page

ofadam, 2021-05-21:
**Describe the bug**
A Database Error Occurred
Error Number: 1055

Expression #2 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'x.countycountconfirmed' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by

select count(distinct COL_CNTY) countycountworked, coalesce(x.countycountconfirmed, 0) countycountconfirmed, thcv.COL_STATE from TABLE_HRD_CONTACTS_V01 thcv left outer join ( select count(distinct COL_CNTY) countycountconfirmed, COL_STATE from TABLE_HRD_CONTACTS_V01 where station_id =1 and COL_DXCC in ('291', '6', '110') and coalesce(COL_CNTY, '') <> '' and COL_BAND != 'SAT' and (col_qsl_rcvd='Y' or col_eqsl_qsl_rcvd='Y') group by COL_STATE order by COL_STATE ) x on thcv.COL_STATE = x.COL_STATE where station_id =1 and COL_DXCC in ('291', '6', '110') and coalesce(COL_CNTY, '') <> '' and COL_BAND != 'SAT' group by thcv.COL_STATE order by thcv.COL_STATE

Filename: models/Counties.php

Line Number: 53

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Awards'
2. Click on 'US Counties'
3. See error

**Desktop (please complete the following information):**
 - OS: OS X
 - Browser: Brave (Chrome)
 - Version: 90
 - PHP 7.4

magicbug, 2021-05-21:
Guessing latest version of Cloudlog? But what version of mysql

ofadam, 2021-05-21:
Latest build of Cloudlog, 8.0.25 of MySQL.

AndreasK79, 2021-05-21:
I'll push a fix for this tomorrow, but I'm pretty sure that changing line 50 in models/Counties.php from group by thcv.COL_STATE to group by thcv.COL_STATE, countycountconfirmed will fix it.

I need to run some tests first to verify, as I don't get these issues here.

ofadam, 2021-05-22:
That fix you mentioned did the trick!

AndreasK79, 2021-05-22:
Excellent. Thanks for testing @ofadam. A fix has been pushed.

PR #1039 (2021-05-23): [Award US counties] A small SQL fix if when sql_mode=only_full_group_…
…by is on. Fixed #1037
commit 164a395b (2021-05-23): [Award US counties] A small SQL fix if when sql_mode=only_full_group_by is on. Fixed #1037
closed 2021-05-23 by magicbug