magicbug/Cloudlog #1044: "Packets out of order" PHP error

ofadam, 2021-05-26:
**Describe the bug**
A PHP Error was encountered
Severity: Warning

Message: Packets out of order. Expected 1 received 0. Packet size=145

Filename: mysqli/mysqli_driver.php

Line Number: 201

**To Reproduce**
Unfortunately, it doesn't happen very often... but when it does, I seem to be:
1. Click on QSO in log to open
2. QSO details appear, but with the error mentioned above showing TWICE.

**Server environment
 - PHP: 7.4.18
 - MySQL: 8.0.25
 - Cloudlog: Latest build

magicbug, 2021-06-04:
this won't be a Cloudlog issue but rather a MySQL configuration one, I'd suggest looking at max_connections settings and making sure things are setup correctly. Googles got quite a bit of info on the subject

magicbug, 2021-06-04:
Also ATTR_EMULATE_PREPARES in the database.php config might help but I think its not possible to set that but could be wrong

closed 2021-06-04 by magicbug