magicbug/Cloudlog #1058: PHP ERROR when viewing QSO without lat & log

M0JGX, 2021-06-07:
PHP ERROR when viewing QSO without lat & log

Undefined Index lat 
Undefined Index long    Lines 441 & 441

Would suggest using isset as a fix

Something like 
if (isset($result['lat'])) {$lat = $result['lat'];}
if (isset($result['long'])) {$lng = $result['long'];}

I'm sure you can come up with something better

[image]

magicbug, 2021-06-07:
Should be resolved

closed 2021-06-07 by magicbug

Jaif, 2022-06-15:
This issue is unfortunately not resolved yet. I also get errors when viewing a certain QSO.
[image]

It seems to also break the dashboard map pointers, the JSON data returned from index.php/logbook/qso_map/25 contains PHP error HTML which makes it invalid as JSON.