magicbug/Cloudlog #1064: LoTW sync broken

ofadam, 2021-06-14:
**Describe the bug**
It appears that the Logbook of the World sync is broken in Cloudlog as of today (it was working yesterday?) The LoTW site now rejects the GET method.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to /lotw
2. Click on the 'manual sync' button
3. See error

**Expected behaviour**
It normally throws back some log details saying how many QSOs have been updated.

**Desktop (please complete the following information):**
Cloudlog version: latest from master build
PHP: 7
MySQL: 8

**Additional context**
A PHP Error was encountered
Severity: Warning

Message: file_get_contents(https://lotw.arrl.org/lotwuser/lotwreport.adi?login=&password=&qso_query=1&qso_qsl='yes'&qso_qsldetail='yes'&qso_mydetail='yes'&qso_qslsince=2021-06-14): failed to open stream: HTTP request failed! HTTP/1.1 405 Method Not Allowed

Filename: controllers/Lotw.php

Line Number: 609

magicbug, 2021-06-14:
Can you check again to see if it's not just a case of lotw being down? Just checked on a few here and seeing no errors

ofadam, 2021-06-14:
It does indeed appear to be back! It must have just been a morning-long outage.

closed 2021-06-14 by ofadam