magicbug/Cloudlog #1066: US county does not populate from Callbook

tarantir, 2021-06-16:
When logging a US callsign the US county does not populate.

1. Ensure Cloudlog callbook is setup (I'm a QRZ Logbook XML subscriber)
2. Enter a us callsign (e.g. N8AUU) on the live QSO screen 
3. Tab to let callbook retrieve the c**Screenshots**
If applicable, add screenshots to help explain your problem.allsign details

**Expected behaviour**
US County will populate on the QSO general tab.

I've attempted to debug this without luck.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

phl0, 2022-04-16:
Thought I'll make a quick fix here due to latest experiences with auto-filling German DOKs. This should be fixed by PR https://github.com/magicbug/Cloudlog/pull/1460

phl0, 2022-05-30:
Its now in station_logbooks.

closed 2022-06-13 by magicbug