magicbug/Cloudlog #1067: Dark mode makes QSO info header on map invisible

mpentler, 2021-06-16:
**Describe the bug**
When using dark mode the tool tips that display when you click on a red dot on the map don’t show the callsign at the top of the bubble. The text is in white, but the background is also white: 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to your display settings and select the dark mode theme
2. Select any red dot from the map and click to open the QSO info
3. See the invisible header. The text can be selected to reveal its presence.

I do have an image of this if required, but not where in posting this from.

**Expected behaviour**
I assume black text instead? Like the rest of the QSO info box.

**Screenshots**
Can post later this evening.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 91.0.4472.106

**Additional context**
N/A

mpentler, 2021-06-18:
Seems it's this that's doing it: 
https://github.com/magicbug/Cloudlog/blob/8ed1141e36fb91e36b3259763afd88dabadb1061/assets/css/cyborg/bootstrap.min.css#L371

There's no corresponding setting of white in, for example, the superhero theme when you inspect the element in developer mode. I think the line can possibly come out?

AndreasK79, 2021-06-18:
You should really put the CSS in overrides.css instead. The problem is that if you comment out that line, the header in gridsquares will disappear. Try putting this in overrides.css in the cyborg-folder instead:
`.leaflet-popup-content h3 {
	color: #000;
}`

AndreasK79, 2021-06-19:
Pushed #1072

commit b4a01660 (2021-06-22): Merge pull request #1072 from AndreasK79/cyborg_css_map_popup_fix

[Cyborg CSS Theme] Fix for callsign in map popup. It was all white. Fixes #1067
closed 2021-06-22 by magicbug