magicbug/Cloudlog #1069: gridtracker and cloud log issue.

lyonrt, 2021-06-18:
Cloudlog interfacing with gridtracker doesn't seem to work.  Gridtracker states "invalid response" when I put in variations of my cloudlog install and the api key.

https://www.ko4nwk.com/cloudlog/index.php/api/ with the api key in the box beneath it.

dg9vh, 2021-06-18:
The DNS name seems to be invalid?

AndreasK79, 2021-06-18:
I just tried with everything running locally here. No problem to log from WSJT-X to Cloudlog via GridTracker.
I use the following url in GridTracker: http://127.0.0.1/cloudlog/index.php/api/qso. Try to add qso at the end of your url

closed 2021-07-03 by magicbug

skymaker-c2is, 2021-07-25:
I'm having the same issue. Thought it was a GridTracker issue, but when I try to manually post a QSO to the API, the server responds with ```401 Unauthorized {"status":"failed","reason":"missing api key"}``` - even though I used a valid API-key.

Here's the payload:
```
{
  "key": "clXXXXXXXXX",
  "station_profile_id": "1",
  "type": "adif",
  "string": "<adif_ver:5>3.1.0<programid:6>WSJT-X<EOH><call:4>TEST <gridsquare:4>JO31 <mode:3>FT8 <rst_sent:0> <rst_rcvd:0> <qso_date:8>20210725 <time_on:6>124811 <qso_date_off:8>20210725 <time_off:6>124811 <band:3>15m <freq:9>21.075500 <station_callsign:6>DO4SKY <my_gridsquare:6>JO31UI <tx_pwr:2>5W <EOR>"
}
```
The error also occurs if I just send it with "string": "test" to the API, so I don't see a JSON-format problem here.