magicbug/Cloudlog #1083: Timeline error for IOTA

ofadam, 2021-07-08:
**Describe the bug**
When looking at the timeline for confirmed IOTA QSOs, I get the error message below.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Analytics and then Timeline
2. Click on checkbox for IOTA
3. Error results

**Screenshots**
A Database Error Occurred
Error Number: 1055

Expression #3 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'cloudlog_kd9nye.iota.Name' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by

select min(date(COL_TIME_ON)) date, col_iota, name, prefix from TABLE_HRD_CONTACTS_V01 thcv join iota on thcv.col_iota = iota.tag where station_id = 2 and col_iota <> '' group by col_iota order by date desc

Filename: models/Timeline_model.php

Line Number: 108

**Additional context**
- PHP 7.4.21
- MySQL 8
- Latest Cloudlog

AndreasK79, 2021-07-09:
@ofadam can you test this fix? Timeline_model.php line 105. Change to `        $sql .= " and col_iota <> '' group by col_iota, name, prefix`

And can you also check if dxcc timeline does work?

ofadam, 2021-07-11:
That fixed things (and DXCC timeline continues to work.)

AndreasK79, 2021-07-11:
Thanks. #1085 pushed now.

magicbug, 2021-07-14:
Merged

closed 2021-07-14 by magicbug