magicbug/Cloudlog #1090: Maritime mobile QSOs

prstoetzer, 2021-07-14:
QSOs with maritime mobile stations are not handled properly by Cloudlog. I recommend adding a "NONE" option for DXCC (DXCC ID of 0) and perhaps automatically selecting the NONE DXCC if the suffix entered after the call is "/MM"

magicbug, 2021-07-28:
I've added "None" as an option in the DXCC dropdowns I need to adjust the callsign lookup bits when you type a callsign in to set that correctly but that's overdue an overhaul.

I'm guessing we arent handling DXCC ID 0 in the awards parts ? @AndreasK79

AndreasK79, 2021-07-28:
@magicbug some places the sql checks for dxcc > 0, and it's also joining on dxcc_entitites. Since dxcc_entities doesn't have an adif = 0, it shouldn't be a problem.

It needs testing to be 100% sure, but it seems that it should work as is.

AndreasK79, 2021-07-28:
@magicbug the summary didn't check for col_dxcc > 0. Fixed that now with a PR.

PR #1116 (2021-07-30): [QSO Entry] Maritime Mobile identification fix
This should fix #1090 by identifying /MM QSOs.

I am not sure if this fixes all use cases regarding /MM. I've fixed the two functions identifying a DXCC.
The REGEX is based on a DXCC Lookup DJ1YFK made: https://git.fkurz.net/dj1yfk/dxcc. I've converted this to PHP, but I'm not sure if we need this complexity in Cloudlog. I ripped out the part for identifying /MM QSOs instead.

Since we have no CQ Zone, I return this as 0. By doing this, I needed to add 0 to dropdown in the QSO Entry.
I also needed to fix the summary in the CQ Zone award to look fo [... cut]
closed 2021-07-30 by magicbug