magicbug/Cloudlog #1106: Security bugs

amammad, 2021-07-27:
Hi dear Cloudlog team.
Can I report vulnerabilities that exist in your nice application?
are you care about security issues ?

dg9vh, 2021-07-27:
Please report it as issue I think that would be the best way... Or Peter
@2m0sql should tell a better way (direct email to him?)

amammad, 2021-07-27:
Oh man, Thanks for your quick response, your the best !!!
In this week I report several vulnerabilities (10 until now!!) in Huntr.dev platform and they send to you a authorization link. OK ?
I and my friend on this platform do bug bounty for open source community. this bug bounty platform also give bounty for fix the vulnerabilities to maintainers !!! 
If  you and your nice team have no problem with these kind of procedure in such good platform, then just tell me to report vulnerabilities there.

magicbug, 2021-07-27:
> Please report it as issue I think that would be the best way... Or Peter @2m0sql should tell a better way (direct email to him?) amammad

Exactly this, this is a small project, any issues will be fixed that are reported.

closed 2021-07-27 by magicbug

amammad, 2021-07-27:
I dont understand what you mean my friend.
can i report them to huntr?

magicbug, 2021-07-27:
If you have a security bug contact me the repo owner directly peter@2m0sql.com 

Please note we don't use bounty programmes or pay any money .

amammad, 2021-07-27:
huntr.dev is a good interactive platform and give very good reputation to each repository as they do it for us absolutely free.

amammad, 2021-07-27:
i don't mean i want to ger feom you money man!!
i'm sorry for my bad

amammad, 2021-07-27:
hutr support all repository in github for free nad give 250$ until 500$ monthly budget for each of ones for making open source world more secure

amammad, 2021-07-27:
Also without any registration🤣
they send you a link that juat for you  and then you can interact with huntr tram or the bug hunter and when you fix your own qpplication they give both of us free bounty.

amammad, 2021-07-27:
just give me some feedback to start working more and more and making good secure application with huntr team and their bug bounty hunters like me :))

magicbug, 2021-07-27:
I've already told you how to report