magicbug/Cloudlog #1110: DXCC List Update not working

scalpya, 2021-07-27:
Hi
I install Cloudlog on my main server where is MarinaDB after some previous problem with some PHP error i thinking everything working fine but when i try Cclick on DXCC update i see only:

Updating...
Dxcc Entities: 0
Dxcc Exceptions: 0
Dxcc Prefixes: 0

and nothing happened. I think its some DB problem ? 

I install Cloudlog on my backup server witht MySQL DB and there is no any problem like this, but backup server is just temporarty so i like to install on my main server.

Thank You

magicbug, 2021-07-27:
Its going to be one of a few things including directory permissions i'd suggest checking Admin -> Debug is all happy, then perhaps check server logs to see why its failing if that doesn't show any issues.

scalpya, 2021-07-27:
Debug show everything green , i check folders now and atribut is 755. i Cant see any error on server,

matthewrwright, 2021-07-28:
Check permissions on the files within /assets/json themselves, the folder may be correct, but the files may not be and may require permission changes. I have made a PR to amend the update script - this may or may not resolve your issue - appreciate your feedback if this solves it

PR #1113 (2021-07-28): Update update_cloudlog.sh
Change to asset directory permission check rather than just QSL

Assets directory as a whole should be permission updated rather than just the qslcard folder, otherwise  https://urlhere/index.php/update/update_dok fails with  

`file_put_contents(./assets/json/dok.txt): failed to open stream: Permission denied`

This is an edge case if the cron script runs as a different user to the www user.

May fix #1110  too ?
closed 2021-07-28 by magicbug