magicbug/Cloudlog #1117: Bearing calculation wrong

dg9vh, 2021-07-30:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Some bearings are calculated wrong, example see screenshot

**To Reproduce**
Steps to reproduce the behaviour:
1. Enter "JN39KF" as locator for station profile
2. Go to New QSO and Enter A61QQ as callsign, also happens with brasilian callsigns or other south-americans or Australian callsigns. 
3. See error

**Expected behaviour**
The right bearing should be shown.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - Windows 10
 - Chrome 64 Bit
 - Version 92.0.4515.107

AndreasK79, 2021-07-31:
Nice catch @dg9vh. Strange that this hasn't been reported before.
The bug is in the qra-library.

I found another bug while looking at it. Try to enter a locator directly to the south or north of you. The distance will not be calculated.

I'm not sure how to fix the bug in the existing code, but I have some code that I can use to replace the existing code.
@magicbug will you take a look, or should I replace the existing code and make a PR?

magicbug, 2021-07-31:
@AndreasK79 I was wondering myself how its not been reported before! its old code but still.. I'm out most of the day so if you fancy correcting it with a PR I'll accept it! else will fix later this afternoon

magicbug, 2021-07-31:
A good example can be found here https://www.dougv.com/2009/07/calculating-the-bearing-and-compass-rose-direction-between-two-latitude-longitude-coordinates-in-php/ with the Great Circle calculation :)

AndreasK79, 2021-07-31:
@magicbug thanks. This was much simpler than the code I had. PR coming up.

magicbug, 2021-07-31:
Merged shall close once I've checked when home but guessing its fine now :)

AndreasK79, 2021-07-31:
@magicbug it should be ok. The tests I did, came up with the same result as another program,

magicbug, 2021-07-31:
Still surprised no one noticed! oh well better late than never

dg9vh, 2021-07-31:
Yesterday was my first time I used a beam :-) all the time before I simply used an endfed wire on shortwave and therefore never been interested in bearings :-) With the beam it was suprising me, why I should beam northward to get to A61 :-)

AndreasK79, 2021-07-31:
I see. Well, thanks to you, two bugs were fixed @dg9vh.

commit 9e058dac (2021-07-31): Merge pull request #1119 from AndreasK79/qra_library_fix

[QRA Library] Fixes #1117. This fixes bearing calculation. Distance w…
commit ea916d88 (2021-07-31): [QRA Library] Fixes #1117. This fixes bearing calculation. Distance was not calculcated if the locator was directly to the north or south of your own locator.
closed 2021-07-31 by magicbug