magicbug/Cloudlog #1118: Trying to get in touch regarding a security issue

zidingz, 2021-07-30:
Hi there,

I couldn't find a `SECURITY.md` in your repository and am not sure how to best contact you privately to disclose a security issue.

Can you add a `SECURITY.md` file with an e-mail to your repository, so that our system can send you the vulnerability details? GitHub suggests that a [security policy](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) is the best way to make sure security issues are responsibly disclosed.

Once you've done that, you should receive an e-mail within the next hour with more info.

Thanks! (cc @huntr-helper)

magicbug, 2021-07-31:
Theres an email in the readme however have created a security policy .md file

closed 2021-07-31 by magicbug

amammad, 2021-08-29:
Hello dear cloudlog team, 
Can I ask you to give me just some feedback about reported issues, please?
Best regards.