magicbug/Cloudlog #1120: Minor Issue DOK P41 missing

M0TZO, 2021-08-01:
Not much more to say. I'm not sure how to edit the file myself, but I can't add DOK **P41** to a QSO as not in list.

Thank in advance

Paul M0TZO

magicbug, 2021-08-01:
Pretty sure dok info comes via a third party list, @phl0 probably point me in the direction of getting it fixed

phl0, 2021-08-01:
The DOK list is pulled from official lists by DARC which is the German club. P41 is not a valid DOK - at least today. See here:

https://www.darc.de/der-club/distrikte/#c261644

I see that we have a problem with logging older QSOs where that may have been a valid DOK though....

dg9vh, 2021-08-01:
We may should add valid from and valid to to the imported data, if darc
delivers that in their database.

Kim

phl0, 2021-08-02:
Apparently they don't provide that information for DOK iirc while they have that for SDOK. I suggest to make the field in the QSO window editable besides having the list of predefined values. This way one could store outdated data which may have been valid for the time the QSO was made.

dg9vh, 2021-08-02:
Well it seems that old DOKs that expired by for example end of existence of the DARC-OV are deleted in the list. So I think @phl0 's idea having the field in that way, that you manually can add values (but for example with an alert notice) which are not in the list... This use-case should mainly only happen when editing or adding past-qsos. On actual "on-line"-QSOs, this should not happen but could, if the DOK list is not actualized on the log-server. So also here this feature could be helpful.

73 Kim

commit 720cf766 (2021-08-12): [DOK autofill fix] Set option create to true so that entries not found in the DOK list, can be manually added. Fixed #1120