magicbug/Cloudlog #1129: State list misses German states

DC7IA, 2021-08-11:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**

The list of state misses German states.

**To Reproduce**
Steps to reproduce the behaviour:
1. open the list
2. be confused :O 

**Expected behaviour**

The list should include the German states! :D

magicbug, 2021-08-11:
Feature rather than a bug, if someone can give me a list of German States compatible with the adif spec we can add it

AndreasK79, 2021-08-13:
@magicbug this needs to be reopened. It was closed by accident, because I put the wrong number in a PR.

magicbug, 2021-08-15:
> @magicbug this needs to be reopened. It was closed by accident, because I put the wrong number in a PR.

no problem issue reopened

Werzi2001, 2021-09-09:
> 
> 
> Feature rather than a bug, if someone can give me a list of German States compatible with the adif spec we can add it

@magicbug I guess this is what you asked for, right?
https://www.adif.org/311/ADIF_311.htm
Table "Enumeration for Country Code 230 (Fed. Rep. Of Germany)"

DL2MF, 2021-11-04:
Use the official ISO codes from this list:

Bundesland | Code

Baden-Württemberg | DE-BW
Bayern | DE-BY
Berlin | DE-BE
Brandenburg | DE-BB
Bremen | DE-HB
Hamburg | DE-HH
Hessen | DE-HE
Mecklenburg-Vorpommern | DE-MV
Niedersachsen | DE-NI
Nordrhein-Westfalen | DE-NW
Rheinland-Pfalz | DE-RP
Saarland | DE-SL
Sachsen | DE-SN
Sachsen-Anhalt | DE-ST
Schleswig-Holstein | DE-SH
Thüringen | DE-TH

The leading DE may be omitted.

DC7IA, 2021-11-04:
> The leading DE may be omitted.

Terrible idea, as for example BY also means Belarus.

dg9vh, 2021-11-04:
How is it done in ADIF specifications?

Werzi2001, 2021-11-05:
> How is it done in ADIF specifications?

@dg9vh I already linked that above:
https://www.adif.org/311/ADIF_311.htm
Enumeration for Country Code 230 (Fed. Rep. Of Germany)
Seems to be the same as the ISO codes.

magicbug, 2021-11-05:
The states field needs to reflect the ADIF type which is this list here https://www.adif.org/311/ADIF_311.htm#Primary_Administrative_Subdivision

If someone wants to build out the system to support it then it would be much appreciated, would need to show them based on the DXCC of the logged callsign.

DL2MF, 2021-11-10:
> 
> 
> > The leading DE may be omitted.
> 
> Terrible idea, as for example BY also means Belarus.

You had a look into the software before posting ... ?

[image]

... otherwise hopefully nobody will log a Delaware QSO ;-)

closed 2024-11-27 by magicbug