magicbug/Cloudlog #1134: after installation wizard tmp is not writable by php

CaptainBalou, 2021-08-17:
Hello,

I just heard aboput this project and tried to test it on my server. I followed the wiki but ended up in an error.

Steps to reproduce the behaviour:
1. Install as shown in the wiki via the installer
2. After installer finished remove install directory as mentioned
3. Navigate to base URL to see cloudlog
4. Error is getting displayed:

An uncaught Exception was encountered

```
Type: Exception
Message: Session: Configured save path '/tmp' is not writable by the PHP process.
Filename: /var/www/clients/client1/web1/cloudlog/system/libraries/Session/drivers/Session_files_driver.php
Line Number: 143
```

**Desktop:**
 - OS: macOS 11.5.2
 - Browser: Firefox
 - Version: 91.0

**Server:**
 - OS: Ubuntu 18.04
 - Webserver: Apache 
 

Can someone assist here?

Thanks in advance
Marco

magicbug, 2021-08-18:
Not so much an issue with Cloudlog but linux setup, you could check the php.ini file to check where the /tmp folder is and its permissions.

Other options edit /application/config.php and find $config['sess_save_path']  and change it to something like

`$config['sess_save_path']          = APPPATH.'writable';`

Then inside the main /application folder create a writable folder and make sure its permissions allows PHP to write to it.

CaptainBalou, 2021-08-19:
Absolutely. I figured out that /tmp was not usable for me so I had to take a tmp inside my hosting area. Working like a charm now. Thanks for your quick answer!

closed 2021-08-19 by CaptainBalou