magicbug/Cloudlog #1146: Feature request: Set all logbook entries as sent to QRZ (like 'mark all as uploaded to eQSL)

philpem, 2021-09-04:
**Describe the bug**
I have imported a lot of old QRZ and eQSL entries and offline logbooks. Some of these were done with ADIF files. When I imported them, they were marked as "not uploaded" for QRZ and eQSL.

I've been able to use the `eQSL Import/Export -> Tools -> Mark all QSOs as sent to eQSL` feature to stop Cloudlog from trying to re-upload these to eQSL, but it still tries to upload them to QRZ.

Would it be possible to add a "mark all QSOs as sent to QRZ" feature which mirrors the function of the similarly named eQSL feature?

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'eQSL', then click the Tools tab
2. Note that there's an option to mark all QSLs as sent to eQSL.
3. Go to 'QRZ'
4. Note that there is no way to do this for QRZ

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser Firefox
 - Version 91.0.2 (64-bit)

philpem, 2021-09-04:
It looks like this is actually there already -- but it's in an unusual place.

If you go to ADIF Import/Export, there's a QRZ tab, which allows you to set them as uploaded.
I'd have expected this to be on the QRZ Import/Export page...!

magicbug, 2021-09-07:
This has been fixed in latest code :)

closed 2021-09-07 by magicbug