magicbug/Cloudlog #1147: eQSL import/export/mark as sent not working

philpem, 2021-09-04:
**Describe the bug**
When uploading QSOs to eQSL, I get a "The following QSOs were sent to eQSL" message suggesting this succeeded- but the QSOs don't seem to have been marked as uploaded.

The "Mark all QSOs as sent to eQSL" feature also doesn't seem to work.

"Import from eQSL" reports that my eQSL username and password are invalid, but the same username and password works to log into the eQSL website.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'eQSL'
2. Click on 'Upload to eQSL'
3. Note that the QSOs are reported as being sent
4. Click on 'Upload to eQSL' again - note that the same set of QSOs is offered for upload again,

1. Go to 'eQSL'
2. Click on 'Tools' then 'Mark QSOs as uploaded to eQSL'
3. Go to 'Upload to eQSL'
4. Note that the list of QSOs to send is not blank -- meaning the QSOs have not been marked as sent.

1. Go to 'eQSL'
2. Click on 'Download from eQSL'
3. Run the download. Note that a "username or password is incorrect" error appears.

**Expected behaviour**
QSOs upload, download and mark-as-sent correctly.

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser Firefox
 - Version 91.0.2 (64-bit)

Werzi2001, 2021-09-13:
@philpem Are you maybe using multiple (attached) accounts on eQSL? I'm doing that for my SOTA acitvations and had the same problem because of that. As my SOTA account on eQSL are using a different call sign (/P suffix) I had to give that as my login name in the Cloudlog settings. Without that I got the same errors as you do.

@magicbug @AndreasK79 Given that this is the reason would it make sense to move the eQSL credentials from the account to the station settings? Or at least the possibility to override the eQSL username for each station (just like the eQSL QTH)?

Werzi2001, 2021-09-13:
@philpem Not sure if I understood correctly. Did that solve your issue or are you still having trouble? Basically the settings in Cloudlog have to exactly match the ones in eQSL. Username has to be the one on the correct eQSL account (not necessarily the same you use for login usually as the attached eQSL account might have something different) and also the "eQSL QTH Nickname" in Clouglog has to match the account nick name given on eQSL. I guess without "QTH nickname" on eQSL it doesn't work either but didn't try.

AndreasK79, 2021-09-13:
> @magicbug @AndreasK79 Given that this is the reason would it make sense to move the eQSL credentials from the account to the station settings? Or at least the possibility to override the eQSL username for each station (just like the eQSL QTH)?

It makes sense to me, but as I do not have or use eQSL, I haven't studied the code around this. It's better that @magicbug pitches in with his thoughts here.

philpem, 2021-09-13:
I've got it working. I had to go into the eQSL "Profile" tab and set the QTH nickname to my callsign -- then update Cloudlog to match.
All the accounts were initially set to a QTH nickname of "Home".... which obviously wasn't going to work.

The confusing part is that when the upload failed, the "status" given was blank. It would be good if the eQSL upload feature gave a more descriptive error message -- like "check your QTH nickname matches the one in the eQSL Profile".

closed 2021-11-04 by magicbug