magicbug/Cloudlog #1148: Feature request: Ability to load call history file to show membership info in log page

m0pcb, 2021-09-04:
Just like in N1MM loading a call history file allows N1MM to show a name and membership number on the log window I would like to be able to upload an N1MM call history file that will show me data about a callsign on the main logging entry page.

Why is this useful? It would highlight a callsign as being a member of your club and show their name or other info about them. Suggest using the N1MM call history format because those files are common and easy to edit/create.

magicbug, 2024-01-01:
[FOCCW.txt](https://github.com/magicbug/Cloudlog/files/13806217/FOCCW.txt)

Attaching the FOC txt from N1MM as example for taking this further

magicbug, 2024-01-10:
OK so i can now read CSVs and get PHP array output similar to this 

[image]

The question really is how to handle this in the frontend..

Does name overwrite that that exists or should this be shown in a box on the right

timkirby, 2024-01-10:
Thanks for looking at this for us!

Good questions! I think one of the things we would like to know is a quick prompt that someone is an FOC (or CWOPs, FISTS etc etc) member as you type the callsign in. Possibly the ability to overwrite the name might be useful. - could do something like insert Tim (CWOps: 2755) into the name field. 

It might be useful to record the membership number in a field so that we could potentially get reports based on the club members that have been worked. Having said that, Fabian's reporting suite is so good -based on an ADIF upload - it's arguable that's not needed at all. I'll be interested to see what Iain thinks.

I also wonder if there is mileage in having the ability to look up more than one call history file. That then raises the question of what you do if someone appears in both files!!

Sorry - I appreciate I have raised more questions than provided answers...

magicbug, 2024-01-10:
Could add the data to the SIG/SIG INFO under General, obviously more tricky if they are a member of more than one group.

timkirby, 2024-01-10:
That looks a good home for the data. How horrid would it be to concatenate the data in that field eg FOC-2218, CWOPS-2755 ? You’d then have to query on the field using a LIKE clause, which I appreciate isn’t ideal!

magicbug, 2024-01-10:
You could do it in the Comments field with FOC:#number# etc then like it out in the search

timkirby, 2024-01-10:
That could work!

magicbug, 2024-01-10:
OK will wait to see what @m0pcb thinks before implementing

timkirby, 2024-01-10:
Great! Thank you.

m0pcb, 2024-01-14:
Apologies for the delay in commenting here.

I reckon it will be good to be able to insert the name into the Name field, ideally with correct character case rather than all caps or all lower case.

Membership numbers would be good to insert into the comments field, as above FOC:nnnn  FISTS:mmmm in a list. There are a variety of different call history files, either club specific or multi-club. It would be great to be able to handle the different types, which the example shows multiple entries.

I might also suggest a pane to the right of the log entry form, maybe above the poane showing callsign matches from the log, which shows club membership, number and name in a more obvious font than reading the log entry form fields.