magicbug/Cloudlog #1149: error Table 'cloudlog.options' doesn't exist

112bzo, 2021-09-05:
Hello is see this after a new install.

A Database Error Occurred
Error Number: 1146

Table 'cloudlog.options' doesn't exist

SELECT * FROM `options` WHERE `autoload` = 'yes'

Filename: C:/xampp/htdocs/system/database/DB_driver.php

Line Number: 691

How to fix this? please help

magicbug, 2021-09-07:
If your seeing this after install its because migration system which creates extra MySQL tables is failing if you go to the /migrate it might give you some idea of whats happening

Werzi2001, 2021-09-09:
@magicbug I get the same error while trying to setup a development environment using docker. Maybe I'm missing some PHP modules but I think I installed everything. Could you please elaborate a bit more on what you mean by "go to the /migrate"? http://localhost/migrate does not exist and so doesnt http://localhost/index.php/migrate. I only see /application/migrations with migration PHP files.

Werzi2001, 2021-09-09:
@AndreasK79 Thanks a lot for your reply. That was the reason. I've been using PHP 8 because the installation on Linux just says "PHP >= 7.0". Now I saw that the main readme states it is not working fully on PHP 8. I tried PHP 7 and it works for me now.

AndreasK79, 2021-09-09:
@Werzi2001 Great that you got it working. I've added a short FAQ now, and also added information that PHP8 is not fully compatible on the Linux page. Hopefully it will help people to avoid this problem.

closed 2021-09-26 by 112bzo