magicbug/Cloudlog #1197: A PHP Error was encountered

112bzo, 2021-09-25:
Hello,

I see this error: 

A PHP Error was encountered
Severity: Notice

Message: Undefined index: qso_date

Filename: models/Logbook_model.php

Line Number: 1566
Line Number: 1569
Line Number: 1914

AndreasK79, 2021-09-25:
Hi.

We need more information to be able to help you out. You need to describe what you did to get this error.

Andreas.

112bzo, 2021-09-26:
i see that when i import a adif file

AndreasK79, 2021-09-26:
Well, is it a certain ADIF-file, or do you get this error with all the ADIF-files you try with?

AndreasK79, 2021-10-06:
@112bzo did you solve this? If not, we need more info to be able to pinpoint the problem.

112bzo, 2021-10-11:
No still have this problem

DL3EL, 2021-12-18:
I do not know,, if that has been solved. However, I currently have the same problem and I did some research. It occures when in the ADIF file the <NAME> contains an Umlaut or similar. In this case the date is set to 01.01.1970. When you expand the qso, you see, that the Umlaut ends with a "<" (Jol< and JRGEN< in the attached QSOs)

<FREQ:9>14.000000<CALL:5>SV0IE<MODE:3>SSB<NAME:6>JÜRGEN<QSO_DATE:8>19940404<QSO_DATE_OFF:8>19940404<TIME_OFF:6>072100<TIME_ON:6>071300<QTH:9>KALAMANTA<RST_RCVD:2>56<RST_SENT:2>59<NOTES:10>via DL2YAG<BAND:3>20m<COUNTRY:6>Greece<DXCC:3>236<QSL_VIA:6>DL2YAG<SRX_STRING:2>56<STX_STRING:2>59<TX_PWR:3>0.0<STATION_CALLSIGN:6>DL3FDU<MY_CITY:9>Frankfurt<EOR>

<FREQ:8>3.713000<CALL:5>F5TBA<MODE:3>SSB<NAME:4>Joël<QSO_DATE:8>19970223<QSO_DATE_OFF:8>19970223<TIME_OFF:6>070300<TIME_ON:6>070300<QTH:6>Amiens<RST_RCVD:2>59<RST_SENT:2>59<BAND:3>80m<COUNTRY:6>France<DXCC:1>0<SRX_STRING:2>59<STX_STRING:2>59<TX_PWR:3>0.0<STATION_CALLSIGN:6>DL3FDU<MY_CITY:9>Frankfurt<EOR>

Hope that helps,
Thomas, DL3EL

AndreasK79, 2021-12-18:
@DL3EL hi Thomas. I copied the 2 QSOs above, added a header to make it an ADIF-file. No trouble importing here.
What kind of errors do you get? Are you running the latest version?

Andreas.

DL3EL, 2021-12-18:
Hi Andreas,
I hope I am running the latest version, I did a git clone yesterday. The ADIF is generated from FLDigi and not changed. The header is:

File: export.adi
<ADIF_VER:5>2.2.7
<PROGRAMID:6>fldigi
<PROGRAMVERSION:6>4.1.20
<EOH>

Sorry, I have not noted the exact error, but it said something, that the field qso_date is missing or empty (which is not). Attached are two screenshot, showing the output:

[image]
[image]

Thomas

DL3EL, 2021-12-19:
I have reduced the adif to import to some recods, which only have names with Umlaute (can I attach a file here?). Each record leads to the following error notice:

[image]
[](url)

Thomas

AndreasK79, 2021-12-21:
@DL3EL You can't attach files here, but I downloaded Fldigi and made an ADIF-file. I get the same error on import. I'll see if I can figure out what is going on.

AndreasK79, 2021-12-27:
@DL3EL the only thing I can see, is that the ADIF generated by FLDIGI is off by one digit: <NAME:5>Joël. Did you edit the ADIF you pasted here? Those seem to have a correct number in the name.

DL3EL, 2021-12-28:
Hi Andreas,

no, I did not edit the export syntactical. I only deleted records to limit it to 4 records for e quicker text. And all these records have the correct number.

 

Some observations: FLDigi is running on my Windows 10 machine, German setup & code pages. Why I export the data and have a look at it with an editor, i.e. Total Commander, it looks good. It uses ANSI coding, “Jürgen” is “4A FC 72 67 65 6E”. In Total Commander I can change to UTF-8, then the display is garbage after the “J”. Depending on the letter, the first one o two letters of the qos tag are missing.

 

When I enter a qso with “Jürgen” into an FLDigi, which is running on a Raspberry, the same Jürgen is exported as “<NAME:7>Jürgen” or “4A C3 BC 72 67 65 6E”. That file copied to my Windows machine and displayed with TC shows UTF-8 as coding on my Windows machine (display looks good).

 

Now the funny part. My logbook on windows has 4 records, all entered on windows. The logbook on the raspberry has 1 entry (from above). When I now merge the exported windows logbook into the existing raspberry logbook, FLDigi on Raspberry shows 5 entries, all umlauts are displayed correct in FLDigi. This logbook again exported contains 5 records, 4 with ANSI and one with UTF-8 coding. The length given in the name tag matches the length of the coding.

 

As the single record does not contain a hint of the coding used and it could be different, I have no idea, how that could be solved

AndreasK79, 2021-12-31:
@DL3EL thanks for the information and the explanation. I'm not sure much can be done to Cloudlog to fix this.

DL3EL, 2021-12-31:
@AndreasK79 if the solution would be, to fix it manually, it would help, if the importer would not only note the line of code, in which the error happened, but also the line in the imported file.

AndreasK79, 2021-12-31:
@DL3EL good idea. I'll make a note of it, and see if I can figure something out.

closed 2022-04-01 by magicbug