magicbug/Cloudlog #1211: Missing "options" DB table, on fresh install.

SmittyHalibut, 2021-10-03:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Following the instructions at https://github.com/magicbug/Cloudlog/wiki/Installation, I get to the end of the Install Wizard where it tells me to `Delete the install directory.`.  When I load the root URL after this, I immediately get:
```
A Database Error Occurred
Error Number: 1146
Table 'cloudlog.options' doesn't exist
SELECT * FROM `options` WHERE `autoload` = 'yes'
Filename: models/Options_model.php
Line Number: 19
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Following instructions at https://github.com/magicbug/Cloudlog/wiki/Installation
2. Get to the end of the Installation Wizard.
3. Load the root URL: https://[my site]/
4. See error

**Expected behaviour**
I'm not sure what's next, but I expected it to ask me to login, or create a new account, or something.

Logging into MySQL, I confirmed there is not a table called `options` in the `cloudlog` database.  Similarly, I don't see `options` anywhere in `install/assets/install.sql` which is where all the other `CREATE TABLE` commands appear to be.

**Screenshots**
Let me know if you need something better than the error text above.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Browser: Firefox, but this sure seems server side.
 - Version: git commit `aa909524a7e7aad5a9efe4fc1ce00a6fb5eb8c95` from 2021-09-27.  Checked out from git 2021-10-03.

**Additional context**
Add any other context about the problem here.

SmittyHalibut, 2021-10-04:
Found the problem, here: https://github-wiki-see.page/m/magicbug/Cloudlog/wiki/error-Table-'cloudlog.options'-doesn't-exist

tl,dr: Use PHP 7.4 instead of 8.0 (which is default in Ubuntu 20.04.)

**WARNING** For Ubuntu 20.04, don't use PHP 7.3 or lower if you're using MySQL 8.0.  The error messages are hard to get to, but the Install process will fail just saying "Can't talk to the database, please check your DB configuration." But if you try to connect to MySQL using PHP directly (eg: see below), you'll see that PHP7.3 and below don't support the `caching_sha2_password` password hashing algorithm that MySQL 8.0 uses by default.  So I had to upgrade to PHP 7.4 to get that support.

How I got this error message:
```
$ php7.3 
<?php
new mysqli('localhost','cloudlog','...deleted...','');
mysqli_connect_errno()
?>
[Ctrl-D]
PHP Warning:  mysqli::__construct(): The server requested authentication method unknown to the client [caching_sha2_password] in Standard input code on line 2
PHP Warning:  mysqli::__construct(): (HY000/2054): The server requested authentication method unknown to the client in Standard input code on line 2
$
```

AndreasK79, 2021-10-06:
If I understand you correctly, you managed to fix all the problems by using PHP7.4? If so, that is great.

Regarding PHP8, Codeigniter needs to be updated (the latest v3 from Github does work), but changes to the code in Cloudlog needs to be done as well. (Removing all the constructors from the models, replace each function, update install script). At least that's what I've identified so far.

AndreasK79, 2021-11-18:
V2 will work with PHP8. It's under development now, so I'm closing this.

closed 2021-11-18 by AndreasK79

SmittyHalibut, 2021-11-18:
An even better solution. Thanks! :-)