magicbug/Cloudlog #1240: CloudLogCatQt Unusable for QSO due to GMT/Local mismatch

garrettben, 2021-10-17:
I'm able to get CloudLogCatQt to upload events to the Radio API without issue, but it looks like the QSO/CAT integration is reading the timestamp of the most recent radio event as GMT rather than local (GMT-4) time.  As a result, the dialog refuses to import the CAT information, complaining that it is timed out. 

[image]

[image]

garrettben, 2021-10-17:
Just to clarify, this is an issue in the Cloudlog QSO integration with the Radio API.  I'm not sure if this happens with other API clients.

garrettben, 2021-10-17:
This may be related to issue #1236 , as it seems to be timestamp-related as well.

garrettben, 2021-10-19:
I stripped out CloudLogCatQt and went with https://github.com/Manawyrm/cloudlog-rigctl-interface and rigctld.

While I had to change the order of the info being sent to the radio API, the timestamps are fine, so I can only assume that the timestamp issue is on the client side.

You can close this issue.

magicbug, 2021-12-27:
This should be resolved now the STATUS now checks based on UTC rather than SQL servers local time