magicbug/Cloudlog #1255: Contest Mode not saving QSO's

JesseRhoads, 2021-10-30:
Start a contest (for example CQ WW SSB) on Contest Page
Enter some QSO's
QSO's will be entered in the local session and displayed on the history there, but are NOT saved in the database.
Once the web page closes, all contacts are lost.
(Good thing I also log on paper first and then add to CloudLog once contact is confirmed)

This is with the latest master / last commit sha 9a578e0
It "used" to work.
I tried updating countries, didn't help.
I can enter contacts in live QSO mode just fine. Just Contest mode that is broken.

JesseRhoads, 2021-10-31:
I thought it might be the recent check for dup code, so I reverted to before that, managed to get one qso to add but nothing after that. Very strange.

JesseRhoads, 2021-10-31:
I've tried debugging this in the browser and code and so am far unsuccessful.

AndreasK79, 2021-10-31:
This is strange. I can't reproduce this here. Any error messages in dev-tools at all?

What PHP version are you running?

olwalter, 2021-10-31:
I came here to report the same issue. I'm on a hosted instance, https://db2ow.cloudlog.co.uk/
If used to have an installation on my Raspi, and if I remember correctly, contest logging works there.

AndreasK79, 2021-10-31:
@olwalter any change you could go to devtools in the browser (F12) -> network -> then click on saveqso and see if it says anything at all in response?

I've tried to reproduce on 3 different instances here now, and all of them work.

JesseRhoads, 2021-10-31:
I get no response data there regardless of if it saved to the DB or not.
Are you saving multiple different QSO's in the same contest session?
Then refresh the page and watch all but the first one disappear.
This is on a pi w/ older OS, php versions are 7.3.29-1~deb10u1

AndreasK79, 2021-10-31:
@JesseRhoads Yep, no trouble there. Just came back and tried now. The QSOs I tried with earlier, and two I logged now.

Two things have been changed lately. One is the dupecheck, but that should not affect the logging.
The other thing is that I've added an option to save the received exchange in the DOK-field.

Before that, I added more exchange options. Do you know if it worked after I added more exchange options?

olwalter, 2021-10-31:
@AndreasK79 Thanks for investigating. I get the attached response. 

When I choose a different Exchange type, "Serial+Exchange", and fill in these fields, saving the QSO works.

No matter what the cause is, maybe it would be good to show the error, so that users see that the insert failed.

A Database Error Occurred

Error Number: 1366

Incorrect integer value: '' for column 'COL_STX' at row 1

INSERT INTO `TABLE_HRD_CONTACTS_V01` (`COL_TIME_ON`, `COL_TIME_OFF`, `COL_CALL`, `COL_BAND`, `COL_BAND_RX`, `COL_FREQ`, `COL_MODE`, `COL_SUBMODE`, `COL_RST_RCVD`, `COL_RST_SENT`, `COL_NAME`, `COL_COMMENT`, `COL_SAT_NAME`, `COL_SAT_MODE`, `COL_COUNTRY`, `COL_QSLSDATE`, `COL_QSLRDATE`, `COL_QSL_SENT`, `COL_QSL_RCVD`, `COL_QSL_SENT_VIA`, `COL_QSL_RCVD_VIA`, `COL_QSL_VIA`, `COL_OPERATOR`, `COL_QTH`, `COL_PROP_MODE`, `COL_IOTA`, `COL_DISTANCE`, `COL_FREQ_RX`, `COL_ANT_AZ`, `COL_ANT_EL`, `COL_A_INDEX`, `COL_AGE`, `COL_TEN_TEN`, `COL_TX_PWR`, `COL_STX`, `COL_SRX`, `COL_STX_STRING`, `COL_SRX_STRING`, `COL_CONTEST_ID`, `COL_NR_BURSTS`, `COL_NR_PINGS`, `COL_MAX_BURSTS`, `COL_K_INDEX`, `COL_SFI`, `COL_RX_PWR`, `COL_LAT`, `COL_LON`, `COL_DXCC`, `COL_CQZ`, `COL_STATE`, `COL_CNTY`, `COL_SOTA_REF`, `COL_SIG`, `COL_SIG_INFO`, `COL_DARC_DOK`, `COL_NOTES`, `station_id`, `COL_MY_GRIDSQUARE`, `COL_QRZCOM_QSO_UPLOAD_STATUS`, `COL_MY_CITY`, `COL_MY_IOTA`, `COL_MY_SOTA_REF`, `COL_STATION_CALLSIGN`, `COL_MY_DXCC`, `COL_MY_COUNTRY`, `COL_MY_CNTY`, `COL_MY_CQ_ZONE`, `COL_MY_ITU_ZONE`, `COL_GRIDSQUARE`) VALUES ('2021-10-31 15:02:14', '2021-10-31 15:02:14', 'A1AAA', '160m', NULL, 1830000, 'CW', NULL, '599', '599', 'test contest', 'test contest', '', '', 'Not Found', '2021-10-31', '2021-10-31', NULL, NULL, NULL, NULL, NULL, 'DB2OW', NULL, '', '', '0', 0, NULL, NULL, NULL, NULL, NULL, NULL, '', '', '', '', 'AL-QSO-PARTY', NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, '', NULL, '', '', '', '', NULL, '1', 'JO62QO', 'N', 'BERLIN', '', '', 'DB2OW', '81', 'GERMANY', '', '14', '28', '')

Filename: models/Logbook_model.php

Line Number: 360

JesseRhoads, 2021-10-31:
OK so here's another data point.
I rolled all the way back to commit 0e810788f5528aada9221048dadc4378afe87255 where the contest mode asks you via a radio button of Exchange Type of Serial or Other. 

I now notice that if I don't fill out all the QSO data, it reproduces the issue; it adds the QSO to the local file / browser table, but does NOT complain about any missing fields, and does NOT save in the database.

I rolled back to the latest and it seems like that is the missing additional data point: all the expected fields have to be entered and correct. 

So my hunch is that there is something silently failing field validation or isn't being validated properly, the app does not notice the db insert fails, and just assumes all is well. 

I tried to crank up all the php logging but I'm not getting any output in any error logs or http responses.
Looks like the other user was able to get one contextual example of the insert error.

JesseRhoads, 2021-10-31:
Also -- even when I do enter all fields and DON'T check the use DOK field, it won't save.
So whatever field mangling that's doing (like setting serial to 0?) masks the bug :)

AndreasK79, 2021-10-31:
Thanks @olwalter and @JesseRhoads. The problem is that the exchange is set to blank, and the database does not accept it. I don't get the error here, so obviously it has something to do with the database settings. 

However this does need a fix.

If one of you could try to change line 27 to 30 in logbook_model to this and let me know if it works:

> $srx_string = $this->input->post('exch_recv') == '' ? null : $this->input->post('exch_recv');
      $stx_string = $this->input->post('exch_sent') == '' ? null : $this->input->post('exch_sent');
      $srx = $this->input->post('exch_serial_r') == '' ? null : $this->input->post('exch_serial_r');
      $stx = $this->input->post('exch_serial_s') == '' ? null : $this->input->post('exch_serial_s');

A better error handling is also needed.

JesseRhoads, 2021-10-31:
Thanks, with that patched it seems to more consistently get entries added into the database.

AndreasK79, 2021-10-31:
I've pushed a fix to the master branch now. Thanks for reporting the issue.

JesseRhoads, 2021-10-31:
I guess what throws me is that even with all the fields input, sometimes it still wouldn't save to the database. 
It makes me think there's some lingering logic where the fields are not going copied as we expect. 
Some trickiness with switching what fields are expected per contest maybe?

AndreasK79, 2021-10-31:
Yeah, it is not impossible that more bugs exists. I'll see if I can implement some error logging, and perhaps also an alert in the frontend when things go bad.

I try to test as much as I can, but sometimes it is difficult to catch everything when different configurations produce different behaviors.

Keep reporting your findings, and we'll get those bugs fixed.

iddq, 2021-10-31:
<p>Type: Error</p>
<p>Message: Call to a member function format() on bool</p>
<p>Filename: /var/www/html/ham/Cloudlog/application/models/Contesting_model.php</p>
<p>Line Number: 154</p>

olwalter, 2021-10-31:
Thank you, @AndreasK79!

iddq, 2021-10-31:
$qso is empty.

iddq, 2021-10-31:
```
<h4>A PHP Error was encountered</h4>

<p>Severity: Notice</p>
<p>Message:  Undefined index: station_id</p>
<p>Filename: models/Logbook_model.php</p>
<p>Line Number: 367</p>
```

AndreasK79, 2021-10-31:
@iddq I take it my installer fix worked?

Can you log normal QSOs?
Did you create a station location?
Did you go to admin and update country files?
Which PHP version are you running?

iddq, 2021-10-31:
Yes, Installer is ok and I've closed that issue, thank you.

I got the following error at create location:

A Database Error Occurred
Error Number: 1366

Incorrect integer value: '' for column `cl`.`station_profile`.`station_dxcc` at row 1

INSERT INTO `station_profile` (`station_profile_name`, `station_gridsquare`, `station_city`, `station_iota`, `station_sota`, `station_sig`, `station_sig_info`, `station_callsign`, `station_dxcc`, `station_country`, `station_cnty`, `station_cq`, `station_itu`, `state`, `eqslqthnickname`, `qrzapikey`, `qrzrealtime`) VALUES ('home', 'JN97MN', 'London', '', '', '', '', 'm0abc', '', '', '', '1', '1', '', '', '', '0')

Filename: models/Stations.php

Line Number: 71

JesseRhoads, 2021-10-31:
@iddq I think you need to download / update countries first from the Admin menu to fix that.

@AndreasK79 I reset again to master and it's "still working" for now.
I see what you mean about some things being due to local environment. :)

iddq, 2021-10-31:
thank you for the help, it works, however it is quite complicated.

AndreasK79, 2021-10-31:
@JesseRhoads yep. And a better installer and/or first setup guide would be nice to have. Anyway, glad it's sorted for now.

@iddq glad you got it to work. We have a wiki with some information, which hopefully should help you along the way. We also have discussions if you get any trouble.

magicbug, 2021-11-01:
Thanks for getting this fixed up, been side tracked with contesting in the last 2 weeks and barely been at home!

closed 2021-11-01 by magicbug