magicbug/Cloudlog #1274: 500 Internal Server Error

DC7IA, 2021-11-04:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**

Updated Cloudlog from a version which was a few weeks old, now I get "500 Internal Server Error".

Old version: 6a42641
New version: cd63704

**To Reproduce**
Steps to reproduce the behaviour:
1. Install an old version of cloudlog
2. update (git pull)
3. 500

**Expected behaviour**
A clear and concise description of what you expected to happen.

200 OK

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

not applicable

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

not applicable

**Additional context**
Add any other context about the problem here.

I don't know anything about PHP, so if you could tell me what to look for I could elaborate. :)

iddq, 2021-11-05:
check your web server logs access.log error.log

DC7IA, 2021-11-05:
Error log contains no error and access log just says "500".

magicbug, 2021-11-05:
500 is internal server error so bit beyond the scope of Github Issues for Cloudlog, as already pointed out suggest checking server logs but that's the limit of my suggestions, you can use Discussions for further discussion on this.

closed 2021-11-05 by magicbug