magicbug/Cloudlog #1275: feature request: prevent to record same qso on multiple submission

iddq, 2021-11-05:
same qso recorded if user pressing the button multiple times or refresh the page in browser with F5

AndreasK79, 2021-11-14:
@iddq I've seen this problem before. Not sure how to fix, other than making a call to the database before inserting.

magicbug, 2021-11-14:
Yes think so, its been a lurking thing in Cloudlog from the start

T012m3n7oR, 2021-11-17:
Should be able to resolve this with an if not exists clause on the insert, I can see if I have time this weekend to review the SQL on how it's inserting.

AndreasK79, 2021-11-18:
Add_qso() in logbook_model is the one inserting. It is called from several places. The one used on import, has a basic duplicate check. The ADIF import has an option to skip for duplicate check, so I think the best would be to move the duplicate check (and extend it to check on satelite and sub-mode) into add_qso, with the option to skip it.