magicbug/Cloudlog #1279: Directory index.php cannot be found

GovFox, 2021-11-08:
Hi, I have made it to step 5 of the tutorial for installing this on Linux Ubuntu. 
5. Run the Cloudlog Install Wizard.

I am on the install Cloudlog page on the Apache web interface at http://192.168.1.190/install/index.php

I have tried filling in different settings but get the same error of 

Directory index.php cannot be found or
Directory /var/www/html/ cannot be found

What am I supposed to put in for the directory field assuming the same directories from the instructions were used? Thanks

GovFox, 2021-11-08:
> 
> 
> As long as all the files are in the root the /var/www/html then yes else no, depends on the server configuration it could well be that its /cloudlog/install

All the files are in /var/www/html but when I try that I just get "Directory /var/www/html cannot be found".
Any tips? Thank you

GovFox, 2021-11-08:
UpdateL original iussue is fixed now I get:

The config configuration file could not be written, please chmod /application/config/config.php file to 777

despite the file already having correct permisson.

schoeke, 2021-12-04:
How did you solve it? I run into a very similar issue. I do not understand what I am supposed to enter into the directory field during the install. Should it be the document root? The install directory? The cloudlog directory (which could be different to the document root)? 

"Just follow the prompts" is unfortunately not very helpful here.

closed 2022-01-14 by magicbug

alexander-vidoni, 2022-05-18:
When i cloned the repo, i did it as 'root'.
So, all files where owned by root:www-data.

my apache root directory was: `/var/www/html/ `for the installation.

After running 

`  chown -R www-data:www-data .` within /var/www/html/

the install process succeeded and two files where created:

`application/config/config.php`
`application/config/database.php`

I left the **directory field** in the installer form blank.

antonov124, 2022-06-02:
same here, i use a proxmox debian ct / i also clone the repo as root.  but the install wiz says also "Directory /var/www/html/index.php cannot be found"  
I Also type chown -R www-data:www-data . /var/www/html/
but the error is still there
@alexander-vidoni vielleicht hast du noch eine Idee? ;)

73 Rainer

phl0, 2022-06-02:
Wiki info enhanced with some info concerning the directory field in the installer.

antonov124, 2022-06-02:
Ok I leave it blank and now : [image]. I use the credentials for my new user

phl0, 2022-06-02:
As discussed on Telegram: You have to have the database user and its permissions set up correctly.