magicbug/Cloudlog #1284: Advanced search seems to fail when adding station_id as a query parameter

ryandzink, 2021-11-14:
I am unable to get search results when I add station_id as a query parameter to the search in the advanced search panel. I have many (60+ and growing) separate station locations for SOTA summits and portable operating and I recently saw I could start searching filtering on station_id. However, when I actually search and specify a station_id the query seems to run indefinitely without returning any results.

**To Reproduce**
Steps to reproduce the behaviour:
Go to Search > Advanced Search
Select station_id and input an existing station ID from the station locations list (e.g. 52)
Click search
The Search button will grey out and spin, but never return results. This behavior seems limited to the station_id field, it does not occur with other query parameters. If other query parameters exist and the search works, but station_id is added to the parameters, the query will fail.

**Expected behaviour**
Search displays all QSOs from that station location

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Firefox
 - Version 93.0

ryandzink, 2021-11-14:
After looking at the code a bit, I commented out this line and got closer to what I was expecting, but I'm sure that has other effects: https://github.com/magicbug/Cloudlog/blob/57f370aa1e20b0b1e30a58cb85b8f177319738ee/application/controllers/Search.php#L383

[image]

With that commented out, the search completes and returns the results I expect, but the search button continues to be disabled and the column headers are in the wrong place.

AndreasK79, 2021-11-14:
@ryandzink thanks for reporting. I know why it fails, but not sure how to fix yet.

The reason why it fails, is because of that join. When column station_id is specified, the table is not specified. The sql query then fails with "Column 'station_id' in where clause is ambiguous".

I added the join for the ADIF export. The export needs certain columns from the station_profile.

I will fix if I can figure out how to do it.

AndreasK79, 2021-11-17:
@ryandzink I just pushed a fix for this bug now.

ryandzink, 2021-11-18:
@AndreasK79 I just updated and the search against station_id works as expected. Thanks!

closed 2021-11-18 by AndreasK79