magicbug/Cloudlog #1303: A problem about import qso from LoTW

BH6AOV, 2021-11-17:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

AndreasK79, 2021-12-04:
@BH6AOV did this work before, or is this the first time trying it?

BH6AOV, 2021-12-05:
> @BH6AOV did this work before, or is this the first time trying it?

This is first time

AndreasK79, 2021-12-05:
@BH6AOV thanks. That confirms what I am thinking. The problem is that COL_LOTW_QSLRDATE is not set at all.
I've pushed a fix with #1323. Could you pull the latest master and give it a try? Please report back if this fixes it.

BH6AOV, 2021-12-05:
> @BH6AOV thanks. That confirms what I am thinking. The problem is that COL_LOTW_QSLRDATE is not set at all.
> I've pushed a fix with #1323. Could you pull the latest master and give it a try? Please report back if this fixes it.

Ok, i will try it and report in here.

AndreasK79, 2021-12-11:
@BH6AOV did it work?

closed 2022-01-17 by AndreasK79