magicbug/Cloudlog #1330: “Gridsquare” and “locator” used interchangeably

mpentler, 2021-12-22:
In a few places these terms are both used, for example in logging a QSO and then viewing a QSO. This could be confusing for some users, as well as just feeling “off” somehow.

Is there any appetite to standardise this throughout the UI? I’m very happy to take this on if it’s simple find/replace work. I’m not very PHP-savvy. What term would be preferred?

Mark/2M0IIG

magicbug, 2021-12-23:
Hi Mark,

Yes its interchanged a bit for standardisation could say that it should be "Gridsquare" 

Peter.

dg9vh, 2021-12-23:
I also vote for Gridsquare...

mpentler, 2021-12-24:
Ok. Leave it with me, will have a go after Christmas Day sometime. I doubt I’ll be able to test it, so it really will be just changing strings.

mpentler, 2021-12-24:
It may only be /application/views/user/profile.php to be honest. We’ll see. If anyone sees one let me know here.

mpentler, 2021-12-24:
Update on this, I’ve identified three places so far and changed those. I can see there are translations to consider too. That may need some extra input. Or Google Translate may be fine…
 https://github.com/mpentler/Cloudlog/tree/gridsquare-string

AndreasK79, 2021-12-27:
@mpentler It seems you got them all, except the one in application/language/english/general_words_lang.php:80

mpentler, 2022-01-05:
> @mpentler It seems you got them all, except the one in application/language/english/general_words_lang.php:80

Thanks! Will sort this one.

Have also taken suggestions for translations - do we want to process that as another PR @magicbug ? May not be able to do em all at the same time.

PR #1376 (2022-01-24): Gridsquare string standardised
Fixes #1330 - or at least partially fixes it - English only for now. More work may be required on the locale side to fix the other languages.
closed 2022-01-24 by magicbug