magicbug/Cloudlog #1331: Summary section on WAS page isn't filtered by mode like states list

ofadam, 2021-12-23:
When selecting a specific mode on the Worked All States page, it will filter the list of states correctly – but won't do the same to the summary below. It would be great to have that summary filtered so that you could see how far along you are on a particular mode.

ofadam, 2021-12-23:
(This also affects the DXCC page and possibly others.)

AndreasK79, 2021-12-26:
You are right. The summary is a total summary, it is not filtered now.

ofadam, 2021-12-26:
Is it preferred for it not to be filtered? I'd be happy to attempt a PR if there's interest in having it filtered instead.

AndreasK79, 2021-12-27:
@ofadam well, I guess it would make more sense that it is filtered the same way. These changes should be done in the station_logbooks branch, so that they will come in v2. We can't merge master into station_logbooks, and there are some changes done in station_logbooks to the awards. If you would like to give it a go, be my guest, if not, I'll make a note of it.

AndreasK79, 2021-12-31:
@ofadam I've added it to the DXCC summary now, but only in the station_logbooks branch. All the filters for the main table are also used in the summary. I'll fix the rest as I have time.

AndreasK79, 2022-01-05:
WAS summary fixed. IOTA and CQ left.

AndreasK79, 2022-01-06:
IOTA and CQ added as well. As mentioned, I've made these changed in station_logbooks. That means that this will be available when v2 releases (or if you are running station_logbooks branch).

closed 2022-01-06 by AndreasK79