magicbug/Cloudlog #1332: Is QSO API changed in v2?

ofadam, 2021-12-26:
When using V2 Cloudlog with Gridtracker, I'm unable to get QSOs to log correctly. My initial thought is that it's because a station location isn't being specified in Gridtracker despite it being part of the API documentation... but maybe it's something else amiss?

AndreasK79, 2021-12-26:
Take a look here: https://github.com/magicbug/Cloudlog/discussions/1131#discussioncomment-1727927

magicbug, 2021-12-27:
Currently under development and subject to change at any moment it will be documented fully once its ready for a decent stable beta test period.

closed 2021-12-27 by magicbug

ofadam, 2021-12-27:
Understood and thank you for all the work on V2! I'm happy to help test (despite my limited skills, sadly.)

AndreasK79, 2021-12-27:
@ofadam we need testers. So any help is welcome. Report whatever you find.