magicbug/Cloudlog #1336: Station County list truncated for state of Missouri

ofadam, 2021-12-27:
I noticed when editing a station location in Cloudlog that the county list for the U.S. state of Missouri ends with "Scott"... when others should come after that in A-Z order (I was looking for Shelby.) Perhaps it's being truncated somehow?

AndreasK79, 2021-12-27:
I think the number of counties are limited in the lookup. I think I fixed this a while back in the QSO edit. I have no trouble finding Shelby there. I'll dig further.

AndreasK79, 2021-12-27:
The limit was set to 100. In QSO edit it was changed to 300. I'll push a fix:
[image]

AndreasK79, 2021-12-27:
This is fixed now in master and station_logbooks.

closed 2021-12-27 by AndreasK79