magicbug/Cloudlog #1345: US Counties filtering

ofadam, 2022-01-03:
On the US Counties page under Awards, I've noticed that things can get messy depending on the source ADIF imports to Cloudloug:

[image]

Some (like Gridtracker) use the "XX, County" format (where XX = state abbreviation.) Other software (like Log4OM) seem to just use the county.

Although I believe the ADIF spec asks for the "XX, County" format, would it be possible to have some filtering happen to basically combine both formats and ignore any upper-case/lower-case differences?

AndreasK79, 2022-01-05:
I see your problem, but I don't know how this can easily be fixed. I followed the ADIF spec when adding auto-complete for counties, and saving/editing them. Anyone having any ideas, please feel free to pitch in.

ofadam, 2022-10-04:
Could some filtering be used before tabulating this stats page?

Would need to do the following:
- Ignore case sensitivity entirely
- Remove "XX," or "XX, " (with space after comma)

That should fix things up, I think?