magicbug/Cloudlog #1346: WAS overlay map doesn't scale on smaller screens

ofadam, 2022-01-03:
When viewing the WAS map in the Awards section, it doesn't auto-scale to keep the whole United States in view on smaller screens. You can scroll, but it might be helpful to see the entire country at a glance (even if smaller.)

[image]

AndreasK79, 2022-01-03:
@ofadam pushed a fix for this now: 
[image]

closed 2022-01-03 by AndreasK79

ofadam, 2022-01-04:
It doesn't look like anything changed on my iPhone with this commit, unfortunately.

AndreasK79, 2022-01-04:
Only tested on pc while changing size of browser. Phone doesn't work I see. Will take another look.

AndreasK79, 2022-01-05:
Needed to add some more html tags:
[image] 
Will push fix later.

AndreasK79, 2022-01-05:
@ofadam Pushed a fix to master now.

commit c7e07896 (2022-01-03): Merge pull request #1349 from AndreasK79/was_responsive_map

[WAS Map] Added tag to make the map responsive. Fixes #1346
commit d852d4eb (2022-01-03): [WAS Map] Added tag to make the map responsive. Fixes #1346
commit a0645996 (2022-01-03): [WAS Map] Added tag to make the map responsive. Fixes #1346

(cherry picked from commit d852d4ebd08489dfbd35b2255242b0b868ef9937)