magicbug/Cloudlog #1353: Contest mode. No duplicate checking?

m0urs, 2022-01-06:
**Describe the bug**

In the wiki it reads for the contest logging: "A basic duplicate check is implemented, which will tell you if the callsign is a duplicate on the same band, mode and contest."

**To Reproduce**

Steps to reproduce the behaviour:
1. Enter contest mode
2. Create an entry
3. Create an entry with same data callsign, band,mode and contest again
4. QSO is just saved in addition without alerting me that this is a duplicate

**Expected behaviour**

I would expect that there is an error popping up either after entering the call sign of a duplicate or at least while saving and a question is asked if you would like to save it even as a duplicate record.

AndreasK79, 2022-01-06:
It's there:
[image]
Nothing is stopping you from working/logging the dupes, it's just informing you that it is a dupe.
The policy should always be to work the dupes. It's less hassle to just work the dupes, than trying to explain that it is a dupe. You also never know if you really are in the log of the other station.

And if you are working S&P, then just don't log it.

I see the dupe check doesn't work for the first QSO. I will make a note of that, but for the following QSOs, it does work.

m0urs, 2022-01-06:
Hm, unfortunately this message does not appear here :-( However, I am using the default theme. I just tried to change it to one of the dark themes but this does not work either.

I am using MS Edge (chrome based) and tried Firefox on Windows as well. Before I log in the start page of Cloudlog appears in the design I chose. After logging in, it changes to the default theme again.

In the developer console of Edge I can see this error message (although I am not sure if this is related):

`Fehler beim Laden der Quell Zuordnung durch devtools: Der Inhalt für https://my.server.com/assets/css/bootstrap-dialog.css.map konnte nicht geladen werden: HTTP Fehler: Statuscode 404,net::ERR_HTTP_RESPONSE_CODE_FAILURE`

m0urs, 2022-01-06:
I just made a "git pull" and it updated some files. After that, the message is now appearing. So, sorry, it was my fault.

However, changing the theme still does not change anything here. I will have a look for that as well tomorrow.

closed 2022-01-06 by m0urs

AndreasK79, 2022-01-07:
@m0urs glad you figured it out. I do see it's slow to show up at times, but that might be my machine, as I'm running it locally when testing. Changing themes should work, but if you can't figure it out, please open another issue.