magicbug/Cloudlog #1358: DB error: Incorrect integer value

peterus, 2022-01-07:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
An error is coming up on a new installed cloudlog (in docker) when creating a station.

**To Reproduce**
Steps to reproduce the behaviour:
1. create a new station and save it

** screenshot **

[image]

looks like the field was not rendered.

**Desktop (please complete the following information):**
 - OS: linux, docker
 - Browser chrome
 - Version ?

**Additional context**
```
A Database Error Occurred
Error Number: 1366

Incorrect integer value: '' for column `cloudlog`.`station_profile`.`station_dxcc` at row 1

INSERT INTO `station_profile` (`station_profile_name`, `station_gridsquare`, `station_city`, `station_iota`, `station_sota`, `station_sig`, `station_sig_info`, `station_callsign`, `station_dxcc`, `station_country`, `station_cnty`, `station_cq`, `station_itu`, `state`, `eqslqthnickname`, `qrzapikey`, `qrzrealtime`) VALUES ('Home QTH', 'JN78DH', 'Linz', '', '', '', '', 'OE5BPA', '', '', '', '15', '28', '', '', '', '0')

Filename: models/Stations.php

Line Number: 71
```

magicbug, 2022-01-07:
Please remember to follow the Wiki you need to go to "Admin -> Update Country Files" and download the latest files.

peterus, 2022-01-09:
this was the "issue" - thanks!

closed 2022-01-09 by peterus

petrkr, 2023-10-04:
I just got this error in latest version too..

Maybe there could be some info about download country files instead crash ?