magicbug/Cloudlog #1360: Error on Distances Worked page

ofadam, 2022-01-10:
When using the latest version of the "station_logbook" branch, I get an "undefined" error in red on the Distances Worked page regardless of the band selected.

AndreasK79, 2022-01-10:
I just noticed this yesterday as well. I pushed a fix. Pull the latest code and report back if it works.

ofadam, 2022-01-10:
Just pulled the latest code and I'm still getting the error.

AndreasK79, 2022-01-10:
Ok. This solved my problem. Can you go in devtools and network. Then click plot. You will se a call to get distances. Click on it, and see if you se an error message.

ofadam, 2022-01-14:
Loads of errors! Here are the first of many: 

`<h4>A PHP Error was encountered</h4>

<p>Severity: Warning</p>
<p>Message:  A non-numeric value encountered</p>
<p>Filename: models/Distances_model.php</p>
<p>Line Number: 70</p>

</div><div style="border:1px solid #990000;padding-left:20px;margin:0 0 10px 0;">

<h4>A PHP Error was encountered</h4>

<p>Severity: Notice</p>
<p>Message:  Undefined offset: 260</p>
<p>Filename: models/Distances_model.php</p>
<p>Line Number: 73</p>

</div><div style="border:1px solid #990000;padding-left:20px;margin:0 0 10px 0;">

<h4>A PHP Error was encountered</h4>

<p>Severity: Notice</p>
<p>Message:  Trying to access array offset on value of type null</p>
<p>Filename: models/Distances_model.php</p>
<p>Line Number: 73</p>

</div><div style="border:1px solid #990000;padding-left:20px;margin:0 0 10px 0;">

<h4>A PHP Error was encountered</h4>

<p>Severity: Notice</p>
<p>Message:  Undefined offset: 260</p>
<p>Filename: models/Distances_model.php</p>
<p>Line Number: 73</p>

</div><div style="border:1px solid #990000;padding-left:20px;margin:0 0 10px 0;">

<h4>A PHP Error was encountered</h4>

<p>Severity: Notice</p>
<p>Message:  Undefined index: count</p>
<p>Filename: models/Distances_model.php</p>
<p>Line Number: 73</p>

</div><div style="border:1px solid #990000;padding-left:20px;margin:0 0 10px 0;">

<h4>A PHP Error was encountered</h4>

<p>Severity: Notice</p>
<p>Message:  Undefined index: callcount</p>
<p>Filename: models/Distances_model.php</p>
<p>Line Number: 75</p>`

AndreasK79, 2022-01-15:
@ofadam thanks. Another bug it seems. I think I managed to reproduce it here. It has to do with 2 or more stations locations, where one of them doesn't have any QSOs on the band chosen. I'll get back to you when I have a fix you can try,

AndreasK79, 2022-01-16:
@ofadam give the latest commit on station_logbooks a go, and report back the result.

ofadam, 2022-01-16:
That did the trick! Thanks so much.

magicbug, 2022-01-16:
@AndreasK79 HF fixed however Sats still broken

AndreasK79, 2022-01-17:
@ofadam great.

@magicbug can't reproduce here. Can you dig up some error messages? I can plot "All" fine here.

magicbug, 2022-01-17:
When I do "SAT" "ALL" and click plot it returns 

{
"Error": "No QSOs found to plot."
}

Which cant be right as there's 11946q in the logbook!

AndreasK79, 2022-01-17:
That doesn't make sense. Though I can't really tell why it's doing this in your case.

That message is returned two places in Distances_model: line 15 and line 59. I'm not sure I can manage to figure this one out without remote debugging, or you take a look yourself.

magicbug, 2022-01-17:
I'll have a poke around see if I can fix it

closed 2022-04-01 by magicbug