magicbug/Cloudlog #1365: Improved support for 33cm band

ryandzink, 2022-01-17:
**Describe the bug**
The 33cm band seems to be handled inconsistently throughout Cloudlog, present in some views but not others, missing from some combo boxes, etc. It's a fairly rare band even here in the US but I have started noticing it now that I have starting logging QSOs with it, but maybe others have noticed the same thing.

I have only been able to input 33cm QSOs via ADIF import so some of these might not be issues if the QSOs were entered directly in via the QSO entry page.

**To Reproduce**

- Places 33cm is not visible/supported:
- QSO entry
- Contest logging
- Gridsquare map
- Distances worked (does not show up even though ADIF import has included QSOs from the band)
- Timeline (does not show up even though ADIF import has included QSOs from the band)
- Accumulated statistics (does not show up even though ADIF import has included QSOs from the band)
- Timeplotter (does not show up even though ADIF import has included QSOs from the band)
- Custom maps (does not show up even though ADIF import has included QSOs from the band)

**Expected behaviour**
33cm should show up in the above scenarios.

I believe I can fix this with a PR (looks like the $bandslots array is missing 33cm in many of the models and that should resolve those) but wanted to document it here before looking deeper.

magicbug, 2022-01-17:
It's probably more that I never added it simple as that, should be possible to add I'd suggest only doing this on the "station_logbooks" branch as it will require changing bits all over the place.

ryandzink, 2022-01-17:
I actually think the same issue applies for the 1.25-meter band and maybe others. I see there is a 1.25-centimeter band listed which might be a mix of the 1.2-centimeter and 1.25-meter bands? The list of bands could easily be something that gets out of hand in the user interface if people don't have equipment to work them, maybe there should be something like the Modes configuration UI to disable/enable bands as well?

magicbug, 2022-01-17:
A universal bands list with freq limits would be very useful I think, least then everything would be populated from the same "knowledge" base

magicbug, 2022-01-17:
On thinking, this would need to be done on a "user" level rather than a global one unless the main "admin" of the install decides for everyone similar to modes

AndreasK79, 2022-01-18:
I think user selectable bands in Cloudlog should be added. It is a major task, so it is something I never started implementing.

I would think a solution similar to this:
1. Table bands - this holds band information
2. Table userxband - or something similar, which keeps active bands for the user
3. Add gui for admin to add/edit/delete bands
4. Add gui to select bands for users
5. Change globally where bands are fetched from

This might be something for the future, but feel free to attack it :)

@ryandzink For now, I think just adding those two bands, would suffice. Use station_logbooks (the dev branch) to make changes. I implemented some changes there, so that many places fetch from models/Bands.php. If I remember correctly, awards and analytics fetch the bands from there. You could try to add the bands in that model, and see if they "magically" appear several places.

The bands are also hardcoded. For QSO entry, they are in the view.

closed 2022-02-12 by AndreasK79