magicbug/Cloudlog #1366: Radio API: allow TX power to be sent

tonyc, 2022-01-18:
For the `api/radio` endpoint, it would be useful to be able to send the transmitter power in watts (or milliwatts maybe?), e.g:

```
{
  "key": "...",
  "radio": "TS-890",
  "frequency": 14055000,
  "mode": "CW",
  "timestamp": "...",
  "power": 95,
}
```

This would update the radio's power as 95 watts. My only other thought would be to be that the power could be interpreted as mW instead, to allow for more fine-grained values.

magicbug, 2022-04-06:
Yes send one over

phl0, 2022-04-09:
With https://github.com/magicbug/Cloudlog/pull/1451 this is implemented now. Works fine with CloudLogCatQt here. Let me know if you have any issues.

tonyc, 2022-04-09:
Amazing! This will go into the next release of open890, whenever I get to it :) 💗💗💗

phl0, 2022-04-09:
Sounds good. Which string is it sending as identifier btw? Maybe I need to tweak Cloudlog a little to also accept power from other software than CloudLogCatQt.

tonyc, 2022-04-09:
The identifier is entirely arbitrary, and it's based on the name that the user enters in for the connection name in open890. It ends up looking like `<name> via open890`

I would assume that Cloudlog would just accept any known parameter, as long as the API key was correct, but I haven't spent much time inside the Clouglog code.

magicbug, 2022-04-09:
As long as the power field is filled it shouldn't make any difference what the rig name is

magicbug, 2022-04-09:
Confirmed this is the case in the code, I'll make this as done

closed 2022-04-09 by magicbug