magicbug/Cloudlog #1368: Radio connection timed-out

on4dgt, 2022-01-19:
I don't know if it's a bug or a misconfiguration somewhere. 

My CloudlogCAT shows: localtime -1
Radio interface shows: localtime -1
datebase entry shows: localtime -1

live QSO shows: Radio connection timed-out: OmniRig 1 data is 60 minutes old.

catcontrol works but the "Radio connection" message won't disapear

pc set to localtime
Linux server set to localtime

magicbug, 2022-01-19:
Can you confirm your running the very latest codebase?

on4dgt, 2022-01-19:
I am Peter.
the cloudlog.sh returns "Already up-to-date"

m0urs, 2022-03-21:
I updated today to the most current version and I now have that message as well. Did not appear before the update.

KE8AKW, 2022-03-23:
I believe this might be a issue with CloudLogCAT. In the bottom part of CloudLogCAT it says something like "Successfully synced with CloudLog @( current time). For whatever reason, it is showing for me 3 hours ahead hence forth why we are getting this data is old error in Cloudlog. Im not sure where CloudLogCAT gets its time from so don't know how to fix. @magicbug

closed 2022-08-27 by magicbug