magicbug/Cloudlog #1381: API Import of ADIF doesn't work in station_logbooks branch

phl0, 2022-01-24:
**Describe the bug**
Using the python script sending WSJT-X log data to Cloudlog doesn't work in V2 / station_logbooks any more.

**To Reproduce**
Steps to reproduce the behaviour:
Send a ADIF QSO data via API to Cloudlog

**Expected behaviour**
The ADIF data should get imported

**Additional context**
Add any other context about the problem here.

Reason for this fail is that Stations->check_station_is_accessible fails because no user session exists. But for API requests there will be no valid user session. So this needs to be skipped on API imports.

AndreasK79, 2022-01-24:
Briefly discussed here: https://github.com/magicbug/Cloudlog/discussions/1131#discussioncomment-1727927

phl0, 2022-01-24:
Jupp, I already talked to @magicbug about this directly. :-)

magicbug, 2022-01-24:
@phl0 with any luck this is now resolved

magicbug, 2022-01-24:
Confirmed fix 👍🏻

commit 96d7a253 (2022-01-24): [API][QSO] Skips check_station_is_accessible() if import is coming from API

Skips check_station_is_accessible() if import is coming from API this is because api qso import doesn't have a session thus automatically fails.

Extra checks must be done on the api calls to make sure things are safe.

Fixes #1381
closed 2022-01-24 by magicbug