magicbug/Cloudlog #1410: Post-QSO won't submit because of a locator issue, but it looks fine

mpentler, 2022-02-25:
I am entering a locator square string into a post-QSO entry, but I am getting the red box around a perfectly ok locator string. Indeed, the map dot looks ok and the distance is calculated ok underneath the locator box entry. It won't let me leave it blank, so I can't submit a QSO right now.

Additionally there seems to be rather a lot of javascript errors. Not really sure which of those you may need.

Just wondering if anyone else had seen something similar.

mpentler, 2022-02-25:
When I click Save QSO, I get this: 

qso:1 An invalid form control with name='cqz' is not focusable.
qso:1 An invalid form control with name='dxcc_id' is not focusable.

magicbug, 2022-02-25:
Is this a new install? is the country files intact? "admin -> update country files", I'm guessing that a Station Locations been created

magicbug, 2022-02-25:
Some screenshots of whats being entered might also be helpful

mpentler, 2022-02-25:
I ran the update country files and got:
Updating...
Dxcc Entities: 0
Dxcc Exceptions: 0
Dxcc Prefixes: 0

Unsure if that is the final output or if the Updating... means it is still running.

mpentler, 2022-02-25:
Image: https://media.discordapp.net/attachments/935938190634778674/946707610927767583/unknown.png

Not a new install, part of the OARC cluster of CloudLogs, so I have been asking the Admin people there about it too and we figured the errors were worth reporting.

magicbug, 2022-02-25:
Unless those numbers go beyond zero and updating changes to done then that will be the issue likely with permissions of the file system

mpentler, 2022-02-25:
Thanks Peter, If it is the permissions will it let me know?

magicbug, 2022-02-25:
I cannot remember without checking the code, if you go to Admin -> Debug there's quite a bit of info there

closed 2022-03-28 by magicbug

mpentler, 2022-04-24:
Just to say I eventually got my instance reset, updating country files worked, and this is all okay now.