magicbug/Cloudlog #1414: Validate Grid Locator Entry Field

ve1cwj, 2022-02-28:
Recently noticed that I had incorrectly entered several locators as "multigrids" without repeating the letters (i.e. DM33,43 vs DM33,DM43). This showed up in the VUCC awards summary table, but took a bit of searching to find the offending QSO. Totally operator error on my part, but perhaps Cloudlog could validate the values entered in this field to ensure they are in the correct format (i.e. XX##, XX##xx, XX##,XX## etc)?

magicbug, 2022-03-28:
This looks like it comes under Pull Request #1445 so hopefully once this is checked and merged it will be resolved

phl0, 2022-03-29:
Yes this would result in an error message being displayed and the QSO not saved. See screenshots of the code:

[image]

[image]

magicbug, 2022-03-29:
Perfect, busy with work this afternoon but will check it later and see if its ok for merge :)

closed 2022-03-31 by magicbug