magicbug/Cloudlog #1424: Ability to automatically or manually send a QSO card image over email?

MikeWills, 2022-03-10:
This would be similar to the functionality within MacLogger DX.

magicbug, 2022-03-28:
Dont use MacLogger to know its exact feature, however if someone wants to define a specification it might well end up getting coded :)

thelovebug, 2022-04-01:
I'm very interested in this one too.  Log4eQSL and Log4OM have been suggested as good solutions, but as I'm a Linux user, neither of them is suitable.

phl0, 2022-04-01:
How would you determine the email address of your QSO partner? I guess not to be asked on air :D

MikeWills, 2022-04-07:
Yes, call book or QRZ.

magicbug, 2022-04-20:
Guessing it just needs to generate an "image" and attach it

phl0, 2022-04-20:
Well yeah. I have that in my mind (but not coded yet) ....

MikeWills, 2022-04-20:
> My thought would be image associated to the station location *uploadable*

Yeah, that's looking like I was thinking.  That's awesome! Sorry, I don't
have the php skills or the time to help on this.

phl0, 2022-04-20:
> Yeah, that's looking like I was thinking. That's awesome! Sorry, I don't have the php skills or the time to help on this.

Nor do I .... :)

phl0, 2022-04-20:
> My thought would be image associated to the station location _uploadable_

I like that idea. Plus some imagemagick to add QSO details to the image (template)?

phl0, 2022-04-20:
Got some more code in my PoC:

[image]

dg9vh, 2022-04-20:
Looks pretty good :-)

dg9vh, 2022-04-21:
As we now have a function in development that could send this emails, now it's time for thinking over the process of sending the emails in general. I think we should talk about two scenarios:

a) live-logging / past-logging
b) importing via ADIF

In a) it would be easy to trigger the sending of the email with saving the QSO after setting a "requested-switch" to have an QSL-Card send for this (maybe set by default?) and set this marker to "sent" after sending the mail.

For b) we would need a cron-job running for example once a day fetching requested QSOs for mailing and then calling the existing function for each QSO in the results.

So we need some additional things:

- a new switch on the QSL-Tab for managing the email-QSL (not requested/requested/sent)
- a cronjob-function that fetches all qsos with requested email-QSLs.
- a functionality to "mass modify" imported adif-QSOs to be marked as "email-QSL-requested".

Any other ideas?

Kim

phl0, 2022-04-22:
Tnx feedback @dg9vh 

Currently the code is executed directly after saving a QSO. There is no GUI for that (yet). The thing that concerns me currently is where to store information about mQSLs being sent or not. I cannot find a suitable ADIF field for this information and QSL_SENT is probably for paper QSLs only. Maybe make a request towards ADIF?! :)

My plan (aligned with @magicbug) is that the image template for the mQSLs should be uploadable in the station location so that you can have a different pic for each station location. This is what I am currently working on.

dg9vh, 2022-04-22:
In ADIF spec there are user-defined and application defined fields. See
http://adif.org.uk/312/ADIF_312.htm#Application_defined_Fields

Maybe we could use them in between... 😉

oh1jw, 2023-08-06:
Hi, I'm throwing in my counter-thought and my alternative solution here.

In my opinion, this is a somewhat questionable feature, because the picture attached to the e-mail does not really confirm the connection when viewed in the award mind. And secondly, couldn't we just send paper cards if cards have to be received and given?

BUT, the bigger issue is that I myself have now started to receive these pictures as an e-mail attachment, for example from my peditions, and the pictures are filling up my inbox for no reason and consuming traffic bandwidth, for example for Märket's (OJ0) weak mobile network.
The postal address is automatically downloaded from the QRZ.com database, but _I have not put it there for these QSL photos._

We already have ready-made services for this kind of image carding, e.g. eQSL - and if you find it difficult to use, please code better, but not so that it sends an email to all your contacts without asking and because it's nice.

I understand that this feature may be perceived as important by some.
Would a compromise option be that the recipient can, if  he/she/they  wishes, download his QSL card if he/she/they deems it necessary?

Just my two cents..

closed 2024-11-27 by magicbug