magicbug/Cloudlog #1437: Feature Request: Contest Logging: Pull info from QRZ.COM

m0urs, 2022-03-20:
I would it find useful, that while doing contest logging, the call sign information like name and gridsquare is also pulled from QRZ.COM as it is done while doing normal logging. Especially I would like to see the gridsquare to see where the operator is located and what the distance is.

If this is not implemented because of performance reasons, maybe there could be a switch to turn this on or off?

Thanks!

magicbug, 2022-03-20:
Not so much a performance issue but it adds an extra layer of code, populating from qrz would make you in the assisted category of most events, therefore we would need to add configuration and it's just not in the scope of this right now

m0urs, 2022-03-20:
Ah, ok. Did not think about that assisted category thing. In that case, maybe there could be a possibility to bulk import QRZ.COM information (especially name and grid locator) later for all selected QSOs? Or an additional option in the logbook for every QSO (like the current option "Lookup on QRZ")?

magicbug, 2022-03-21:
You can run index.php/update/check_missing_grid which will go grab grid squares off callbook services, I don't think it will do names.

m0urs, 2022-03-21:
Ah ok. Great, I missed that. Grid would be the most important. I will give
it a try.

OE2WNL, 2022-03-27:
in the class:
/application/libraries/Qrz.php
on line 57 the third parameter is expected ($use_fullname)
`public function search($callsign, $key, $use_fullname)`

I changed this to:
`public function search($callsign, $key, $use_fullname = true)`
to set a default value for using the fullname from qrz response

now it works.

Can @magicbug  repair this or should I make a PullRequest?

Greetings de Wolfgang, OE2WNL

m0urs, 2022-03-27:
Thank you, Wolfgang. Worked for me as well. Was just in time to get the infos to my over 100 contest QSOs today :-)

magicbug, 2022-03-28:
> Can @magicbug repair this or should I make a PullRequest?

Pull request never refused!

OE2WNL, 2022-03-29:
> > Can @magicbug repair this or should I make a PullRequest?
> 
> Pull request never refused!
And here we go:
https://github.com/magicbug/Cloudlog/pull/1446

closed 2022-04-01 by magicbug